EnviroDIY/Arduino-SDI-12 #15: Has anybody worked on porting this to 32-bit boards?

rkertesz, 2016-04-28:
Before I jump in and try to figure out reworking this to avoid avr/interrupt.h and avr/parity.h, I would like to ask if anybody else has been working on this. It would take me a long time to accomplish this, given my total unfamiliarity with how these interrupts function. I am used to simply using attachInterrupt() on my 32 bit board.

Kevin-M-Smith, 2016-04-28:
Hi Ruben - 

I haven't done any work porting to 32-bit, but I might be able to investigate the plausibility if you have a particular board in mind.
 
At one point I looked into porting it to ARM CPUs (e.g. Arduino Due) but the interrupts are handled much differently, especially if there is a bona fide operating system on board.

Best,
Kevin

rkertesz, 2016-04-29:
Thanks Kevin,

I have a couple of konekt dash (cortex m4) boards. I may be able to send one to you to test with. I also have a particle electron (cortex m3). One interesting thing about the particle board (no pun intended) is that someone made some sort of interrupt header file as part of a larger port to particle . https://community.particle.io/t/implementing-timer-interrupts-for-pulse-sensor-solved/2997/14

Both allow for someone to use attach_interrupts. I presume this is not a practical option. 

Thank you,
Ruben 

> On Apr 28, 2016, at 5:44 PM, Kevin M. Smith <notifications@github.com> wrote:
> 
> be able to investigate the plausibility if you have a particular board in mind.
> 
> At one point I looked into porting

Kevin-M-Smith, 2016-05-06:
Hi @rkertesz - 

Apologies for the radio silence.

I think your original assessment was correct: the way these CPUs handle interrupts is fundamentally different from the Atmel ATMEGA series chips.

Unfortunately, I don't think this library will get you there, unless you want to spend one of your M3 or M4's UARTS to piggyback off of a separate ATMEGA chip running the library. 

You would only need one per installation since the address space is enough to handle 64 sensors. 

Did you come up with anything on your end?

Best,
Kevin

ManuelJimenezBuendia, 2016-12-01:
Hi Kevin.

Thank you so much for the library, it works great with AVR based Arduino Boards. Now I am moving to ARM boards (Arduino Zero) and I am having the same issue with avr/interrupt.h and avr/parity.h.
I was wondering if you are planning to modify the library code to make it work with ARM processors. Probably it is not so complicated, but I don't feel as experienced at programming as to do it myself.

Best,
Manuel.

Kevin-M-Smith, 2016-12-01:
Hi Manuel,

Unfortunately the switch from AVR to ARM is not trivial, but @jrzondagh mentioned in issue #13 that he had managed to get a solution working. As far as I know the solution has not yet been made public. 

Warm regards,
Kevin

jrzondagh, 2016-12-01:
Hi All,

Sorry for the delay in submitting the code online, will try get to it this week still. We have it working nicely on SAMD21 chip.

Cheers, Justin

Kevin-M-Smith, 2016-12-01:
That's excellent news. Thanks Justin!

ManuelJimenezBuendia, 2016-12-02:
That's good news indeed. I'm looking forward to testing it. Thanks Justin!

adnanoner, 2017-01-05:
@jrzondagh Hi Justin, great that you are working on this. Any updates :)?

gppk, 2017-01-06:
Hi @jrzondagh How are you getting on with the ARM conversion?

pirvanc, 2017-01-18:
@jrzondagh , Hi Justing, we're also interested in this. Thanks.

locomundo, 2017-01-18:
Yes, please!

ManuelJimenezBuendia, 2017-01-19:
I have modified the code to work with ARM (implemented the parity function and modified the interupt calls) but I am having issues with the Arduino Zero and the GPIO pin that uses the interrupt. The modified code works OK with UNO and Leonardo (both AVR) but fails with Zero.
If anyone wants to have a look at the modified code and the issues, I can send it by email.

adnanoner, 2017-01-19:
@ManuelJimenezBuendia sure, you could send it to adnanoner  ﹫  gmail.com 👍

madvoid, 2017-01-31:
Hi everyone, I've gotten this library working with an Arduino Zero Clone ([this](https://www.sparkfun.com/products/13672) one specifically) and a Decagon anemometer that communicates via SDI12. I made very minimal changes to the library but I don't know if it would be ready to push yet. I've included the files below but I think it would be best if there was more testing with other sensors to make sure I didn't accidentally make it just work with my sensor. 

The changes I made:
* Replace the parity function with my own. It's probably not as efficient as the built in one but the ARM boards are faster so I figure there's not much of a problem.
* Replace the interrupt code in setState() to be compatible with an Arduino Zero. Since the Zero is interrupt compatible on any pin (except for 4), the changes were relatively simple.

Enjoy!

[SDI12_Zero.zip](https://github.com/EnviroDIY/Arduino-SDI-12/files/741537/SDI12_Zero.zip)

Kevin-M-Smith, 2017-01-31:
Thanks @madvoid!

adnanoner, 2017-02-01:
Thanks @madvoid! I got this working (by coincidence I also have the SAMD21 board). For some reason I think there is a problem with flushing or something. The code does however works properly when I begin and end the serial communication after each command 👍

ManuelJimenezBuendia, 2017-02-01:
Hi, a got it working with SDI12_Zero library, but placing forcehold()
before sending the command!! I attach an example code.
There is some issue with setting the pin as output after having attached
the interrupt.

-- 
Dr. Manuel Jimenez
---------------------------------
Universidad Politecnica de Cartagena / Technical University of Cartagena
Departamento de Tecnologia Electronica
Campus Muralla del Mar, s/n
30202 Cartagena. Spain.
Pho: +34 968 338888 Fax: +34 968 325345
e-mail: manuel.jimenez@upct.es
URL: http://www.dte.upct.es / http://www.dsie.upct.es

madvoid, 2017-02-01:
I'm glad a few of you have gotten it working! I have been in email contact with @ManuelJimenezBuendia and I was interested to hear he had the exact same bug as me. I'll describe it now for future reference, and our solutions as well.

The problem we both had was odd. The sensor would respond after the first request for data, but none of the subsequent requests. For both of our sensors, it seemed that digital write didn't seem to change the pin state anymore after a request has been made. 

It seems that @ManuelJimenezBuendia fixed it by issuing a forceHold() command before any request for data, I fixed it by issuing the forceHold() command after every request and response. It seems that the forceHold() is necessary to get the line into a state that's "open" again. @adnanoner, did you need to do this as well?

Finally, I would like to note that in my application, I was using a level shifter that had pullup resistors on the data line. I don't know if these are necessary or not.

ManuelJimenezBuendia, 2017-02-01:
Hi!
I finally made it work! Just by placing a pinMode(_dataPin,INPUT); before every pinMode(_dataPin,OUTPUT);
I don't know why, but this way the pin changes state as output. There is probably some issue regarding pinMode and interrupts.
If anybody finds an explanation, please share.
Meanwhyle, here is the modified library with some working examples. I've used Decagon 5TE powered and operated at 3.3 V (that is directly connected to a Genuino Zero).
[SDI12_ARM.zip](https://github.com/EnviroDIY/Arduino-SDI-12/files/745447/SDI12_ARM.zip)

ManuelJimenezBuendia, 2017-02-01:
I have forked the original project with the [modified versions](https://github.com/ManuelJimenezBuendia/Arduino-SDI-12) of the files to work with Zero and compatible boards:
Hope no have made any mistake!

SRGDamia1, 2017-05-16:
Pulled in and tested on an Adafruit Feather M0. See #33

closed 2017-05-16 by SRGDamia1