EnviroDIY/Arduino-SDI-12 #17: Help with Using library

lilcano, 2017-01-25:
Hello 
I am new to Arduino and the SDI12 based sensors, I am currently working on a project to transfer data from SDI12 based sensors (Decagon 5TM sensors) wirelessly with help of the Arduino and Xbee. Can you please help me with a basic code to collect data from this sensor and display them on the Arduino serial monitor.

Kevin-M-Smith, 2017-01-26:
Hi @lilcano,

I'd recommend working through the examples if you have not already done so: 
https://github.com/EnviroDIY/Arduino-SDI-12/tree/master/examples

__Please get back in touch after you've tried those and we'll try to answer any questions you have.__

## I'm assuming your probe is wired up like this: 
[image]
https://github.com/Kevin-M-Smith/SDI-12-Circuit-Diagrams/blob/master/basic_setup_usb.png

Kevin-M-Smith, 2017-01-31:
@lilcano, did this solve your issue?

lilcano, 2017-03-14:
hello @Kevin-M-Smith 
i got the decagon sensors communicate with the arduino with the help of the SDI12 library. Planning on moving to logging the data to an SD card with the help of the data logging shield from adafruit. having some issues with getting the data from the sensor and storing it on the SD card. Have you done similar task or do you have any resources that can be of help? Thanks

SRGDamia1, 2017-03-14:
We've logged data from decagon loggers to an SD card via SDI-12 without trouble, though I don't have a simple example to give you.  Can you explain where you're getting fouled up?

lilcano, 2017-03-16:
@SRGDamia1 

Can actually see results from the sensor being stored on the SD card when i run my sketch on the arduino. Did you guys use Arduino and a datalogging shield, if yes which shield?. In my case i am using the data logging shield from adafruit. Thanks

SRGDamia1, 2017-03-16:
@lilcano The SD card and shield shouldn't make any difference at all to the functionality SDI-12 library.  If you can get data from the SDI-12 device, the library is working.

You need to send your device a M! command to get it to take a reading, wait the required amount of time, then send the D0! command and parse the results into a variable or buffer.  Then print that variable or buffer to the SD card.

The SD card library and the SDI-12 libraries are fairly RAM heavy, in Arduino terms.  It's possible that you are maxing out your board's RAM and causing it to crash.  I think if that's happening, though, you'd see more obvious problems, like the board restarting itself.

Maybe later this week I can write up a simpler example for you, but it might be faster if you share your setup and explain in more detail the problem your having.

lilcano, 2017-03-24:
Please can you take a look at my sketch bellow and let me know why i am having different results from the decagon 5TM sensors. Running three sensors together and all three are in water in the same bowl. Attached is the sketch and sample results gotten from the sensors. Thanks 

[uno.txt](https://github.com/EnviroDIY/Arduino-SDI-12/files/866683/uno.txt)

[sketch.txt](https://github.com/EnviroDIY/Arduino-SDI-12/files/866686/sketch.txt)

SRGDamia1, 2017-03-24:
The differences between the sensors does not appear to have anything to do with your program.  It is not a significant difference.  If you look more closely at the results, while not identical, they are all within the accuracy specifications of the manufacturer.  (See: http://www.decagon.com/en/soils/volumetric-water-content-sensors/5tm-vwc-temp/)  All sensors show Ea within 15% of the mean value, VWC within 3% of the mean value, and temperature within 1° if the mean value.  

[5TM.xlsx](https://github.com/EnviroDIY/Arduino-SDI-12/files/868569/5TM.xlsx)

closed 2017-03-24 by SRGDamia1

lilcano, 2017-03-25:
@SRGDamia1 

Thanks for the clarification. have you done any work on wireless transmission of the data through Zigbee? if yes can you please provide me with a sample code to get the transmission going want to use one of Zigbee antennas as coordinator(connected to a laptop) and the other as a router (on arduino). was wondering is it better to store the data to the SD card before transmitting the file from the SD card or to transmit while storing at the same time? 

Thanks 
Bauldwin