EnviroDIY/Arduino-SDI-12 #28: inconsistent results

lilcano, 2017-04-23:
Hello, 

I have used this library to communicate and get results from an SDI-12 based sensor(Decagon 5TM). after collecting Ea and VWC data for about 4 days i decided to check the data and noticed the results are not consistent. as shown bellow in the plot. has any one had such results? please help.
[image]

SRGDamia1, 2017-04-24:
What do you think is the problem with your data?  Or, rather, where was your sensor located, which variable were you logging, exactly how were you logging it, and what result did you expect?

SRGDamia1, 2017-04-24:
This is not an issue with the SDI-12 sensor library.  Please do not post questions about your data as GitHub issues with the library.  You can continue to ask in this forum question:  https://envirodiy.org/topic/inconsistent-results/

closed 2017-04-24 by SRGDamia1