EnviroDIY/Arduino-SDI-12 #35: SDI-12 Transparent Mode

AshuJoshi, 2017-05-21:
Does this library support the "transparent mode"? I have an Enviropro Probe that I am trying to use with this library but so far no luck in getting any response back from the probe.

Kevin-M-Smith, 2017-05-21:
Hi @AshuJoshi  - as far as I understand it, "transparent mode" just means that it looks like your computer is speaking directly to the sensor when in fact it is talking to the data logger _transparently_. 

See `examples/g_terminal_window/g_terminal_window.ino` for an example of how to do this.

AshuJoshi, 2017-05-21:
Thanks - tried that but basically I did not get any response back. I am basically new to SDI-12. This is the probe that I am using -[EP100](http://www.enviroprosoilprobes.com/Pages/About/#variants) 
Here is the [SW manual](http://www.enviroprosoilprobes.com/Files/EnviroPro_SDI-12_Commands.pdf) for the probe.

I am using an Arduino Uno. The probe needs supply voltage in the range 6V to 15V - I am powering the probe via a separate DC power supply at 12V - connected the red and black to the power supply.

The blue - or the data wire - I have tried connecting with different data pins - 3, 8, 9, 12 but no change in behavior.

SRGDamia1, 2017-05-22:
Reading what I can from the Enviropro documentation, it looks like it's the probe itself that needs to be in "transparent" mode to talk to a sensor with standard SDI-12 commands.  I found instructions in the EnviroPro Configuration Unit (EPCU) manual (http://www.enviroprosoilprobes.com/Files/EnviroPro_EPCU_User_ManualV2.pdf) on how to put the sensor into transparent mode (see page 14), but I can't find any manual for the sensor itself or any instructions on how to get into transparent mode without the EPCU device.  I suggest you contact EnviroPro directly to ask them.

I also suggest you try posting in the Envionmental Sensors form at envirodiy.com (https://envirodiy.org/forum/environmental-sensors/) to see if you can get a response from someone familiar with this particular probe.

I am going to close this issue because it's sensor specific and not actually a problem with the library itself.  If you do find that there's something about the library itself causing the problem, please reopen the issue.

closed 2017-05-22 by SRGDamia1