EnviroDIY/Arduino-SDI-12 #37: Issue on compilation // Sodaq Explorer - 32 Bits - Cortex M0+

PChignon, 2017-06-15:
Hi,

I don't know if you're library is compatible with Cortex M0+ (SAMD21J18), it's board with  LoRa modem (Low Band), and bluetooth.
In first time, when i selected this type of board, the IDE put your library in not compatible directory, when i take a sample and i try to compile, i have this errors:

Compilation du croquis...
"/Users/pascalchignon/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10802 -DARDUINO_SODAQ_EXPLORER -DARDUINO_ARCH_SAMD  -D__SAMD21J18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="SODAQ"' '-DUSB_PRODUCT="SODAQ ExpLoRer"' "-I/Users/pascalchignon/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/Users/pascalchignon/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/Users/pascalchignon/Library/Arduino15/packages/SODAQ/hardware/samd/1.6.14/cores/arduino" "-I/Users/pascalchignon/Library/Arduino15/packages/SODAQ/hardware/samd/1.6.14/variants/sodaq_explorer" "-I/Users/pascalchignon/Documents/Arduino/libraries/SDI12" "/var/folders/b_/fzlgd0l12f79w7x8vsjv411w0000gn/T/arduino_build_659337/sketch/e_simple_parsing.ino.cpp" -o "/var/folders/b_/fzlgd0l12f79w7x8vsjv411w0000gn/T/arduino_build_659337/sketch/e_simple_parsing.ino.cpp.o"
In file included from /Users/pascalchignon/Documents/Arduino/libraries/Arduino-SDI-12-develop/examples/e_simple_parsing/e_simple_parsing.ino:57:0:
/Users/pascalchignon/Documents/Arduino/libraries/SDI12/SDI12.h:63:5: error: 'KINETISK_UART_t' does not name a type
     KINETISK_UART_t   *REG;
     ^
/Users/pascalchignon/Documents/Arduino/libraries/SDI12/SDI12.h: In constructor 'UART::UART()':
/Users/pascalchignon/Documents/Arduino/libraries/SDI12/SDI12.h:44:9: error: class 'UART' does not have any field named 'REG'
         REG( nullptr ),
         ^
/Users/pascalchignon/Documents/Arduino/libraries/Arduino-SDI-12-develop/examples/e_simple_parsing/e_simple_parsing.ino: At global scope:
e_simple_parsing:60: error: invalid conversion from 'int' to 'Stream*' [-fpermissive]
 SDI12 mySDI12(DATAPIN);
                      ^
In file included from /Users/pascalchignon/Documents/Arduino/libraries/Arduino-SDI-12-develop/examples/e_simple_parsing/e_simple_parsing.ino:57:0:
/Users/pascalchignon/Documents/Arduino/libraries/SDI12/SDI12.h:90:5: error:   initializing argument 1 of 'SDI12::SDI12(Stream*)' [-fpermissive]
     SDI12( Stream *port ) {
     ^
If you hava an idea, of the problem, i need to do a POC, read a Aquacheck 4 probes ( 2 xTemp & 2 x humidity), and transfer this informations with the modem.

Regards,

SRGDamia1, 2017-06-22:
I'm sorry for not getting to this yet.  I don't have a Sodaq Explorer, so I can only get as far as compiling for the board.

I am not getting any compilation errors for the SODAQ ExpLoRer in the Arduino.cc IDE when I use SODAQ SAMD Boards version 1.6.11 (which is the most recent version accessible in the boards manager).  Is that the IDE and the board version you are using?

I am getting errors when I try to use PlatformIO to compile, but not the same errors you are seeing.  If it is PlatformIO that's giving you trouble, try the Arduino IDE.

SRGDamia1, 2017-06-26:
Just FYI, I raised an issue on the PlatformIO side, and it sounds like it is in the process of being fixed.  (https://github.com/platformio/platformio-pkg-framework-arduinosam/issues/2)

If your issue was with the Arduino.cc IDE, I'm kind-of stumped.

SRGDamia1, 2017-06-28:
PlatformIO has been updated, fixing that issue and the Arduino.cc IDE also succeeds, so I'm closing this issue.  If you're still having trouble, let me know and I can reopen this.

closed 2017-06-28 by SRGDamia1