EnviroDIY/Arduino-SDI-12 #47: Compatibility with ESP8266

matiasAS, 2018-07-12:
Hello,
I would like to know if it already exists, there are plans to develop it with an estimated date of publication, the compatibility of this library with the ESP8266 chip

Thank you
regards
Matias Alvarez Sabate

SRGDamia1, 2018-07-22:
I'm sorry, but, no we don't have any current plans to adapt this library for the ESP8266 (or ESP32).  I would be happy to review any pull requests to add the board, though.

PTS93, 2019-02-20:
In version 1.1.0 it was pretty easy to adapt it. Adding
```
#if defined(ESP32) || defined(ESP8266)
enum LookaheadMode{
    SKIP_ALL,       // All invalid characters are ignored.
    SKIP_NONE,      // Nothing is skipped, and the stream is not touched unless the first waiting character is valid.
    SKIP_WHITESPACE // Only tabs, spaces, line feeds & carriage returns are skipped.
};
#endif
```
to SDI12.h was enough to make it work. It was just a thing of LookaheadMode not being supported by those plattforms.

In the newer versions you would have to define those timers the library now uses. Shouldn't be too hard to figure out either.

SRGDamia1, 2019-02-26:
It's on my to do list to re-read the timers on faster boards. Almost all my testing has been on a 8MHz board and at that speed, to get the reception to work fast enough you need to directly check the timer registers. I suspect with faster boards you could just call micros () and not set any timers directly.

On my to do list to test out and implement, but not likely to happen right away.

falives, 2019-11-07:
I have a ESP32 and I have been gradually working on for the past year reading various sensors with the aim of eventually reading SDI12 sensors.  I am stunned to learn that ESP32 can not read SDI12.  PTS93 indicated if I use release 1.1.0 and modify it it can work.  I am not sure if it will still work, how to do it and any other options developed since?

mcca6248, 2020-02-03:
Regarding the issue with ESP32 reading SDI-12 sensors - I am also trying to edit the library as PTS93 suggested, but I cannot find where to download the SDI-12 Library v 1.1.0, any suggestions or developments on this problem?

Thanks!

SRGDamia1, 2020-02-03:
https://github.com/EnviroDIY/Arduino-SDI-12/tree/v1.1.0

jnogues, 2020-03-13:
This patch work in the last release?

protttoy, 2020-07-02:
Anyone had any luck running the library on ESP32/8266 ?