EnviroDIY/Arduino-SDI-12 #53: Lockup on SAMD Based Arduino MKR 1500

bleckers, 2019-07-30:
I am experiencing a lockup issue with the operation of this library on the newer MKR 1500 (ATSAMD21G18 based). 
This happens with all the verbatim examples and current version of the library when you try to do any SDI-12 operations (sdi.sendCommand for example).

At first I thought it might be clock gen/TC conflicts, but running the configSDI12TimerPrescale on it's own works just fine. Then I thought it might be the USB CDC interface being problematic with the way the global interrupts are being disabled, but disabling all Serial operations (just had an LED set up to blink after each operation) doesn't stop the lockups.

The differences between the Zero and MKR 1500 are here in the variants files (left is Zero, right is MKR1500) - variant.h - https://www.diffchecker.com/3s0SIx1t , variant.cpp - https://www.diffchecker.com/iWa7l4fq

These are the same files as the ones in here - https://github.com/arduino/ArduinoCore-samd/tree/master/variants

Is there anything that I am missing?

bleckers, 2019-07-30:
Turns out it's something up with passing the object in that example.

If you change the example to pass the object as a pointer (and use the different begin function):

     SDI12 mySDI12;

      pinMode(sensorPins[pin], INPUT);
      mySDI12.begin(sensorPins[pin]);
      Serial.print("Checking pin ");
      Serial.print(sensorPins[pin]);
      Serial.println("...");
      scanAddressSpace(&mySDI12);
      mySDI12.end();

And also change the supporting functions to pass in as pointer (e.g. void printInfo(SDI12 * sdi, char i){) as well as how you reference the sdi object (from . to ->) it works just fine and dandy.

There must be something up possibly with the newer versions of the compiler.

bleckers, 2019-07-30:
Interestingly when I connect the probe to a pin, the Arduino locks up with this code and outputs a ~48MHz signal on that pin.

Here's the example code:

```
/*
  ########################
  #        OVERVIEW      #
  ########################

  Example C: Checks all addresses for active sensors, and prints their status to the serial port.

  This is a simple demonstration of the SDI-12 library for Arduino.

  It discovers the address of all sensors active and attached to the board.
  THIS CAN BE *REALLY* SLOW TO RUN!!!

  Each sensor should have a unique address already - if not, multiple sensors may respond simultaenously
  to the same request and the output will not be readable by the Arduino.

  To address a sensor, please see Example B: b_address_change.ino

  #########################
  #      THE CIRCUIT      #
  #########################

  You  may use one or more pre-adressed sensors.

  See:
  https://raw.github.com/Kevin-M-Smith/SDI-12-Circuit-Diagrams/master/basic_setup_usb_multiple_sensors.png
  or
  https://raw.github.com/Kevin-M-Smith/SDI-12-Circuit-Diagrams/master/compat_setup_usb_multiple_sensors.png

  ###########################
  #      COMPATIBILITY      #
  ###########################

  This library requires the use of pin change interrupts (PCINT).
  Not all Arduino boards have the same pin capabilities.
  The known compatibile pins for common variants are shown below.

  Arduino Uno:   All pins.

  Arduino Mega or Mega 2560:
  10, 11, 12, 13, 14, 15, 50, 51, 52, 53, A8 (62),
  A9 (63), A10 (64), A11 (65), A12 (66), A13 (67), A14 (68), A15 (69).

  Arduino Leonardo:
  8, 9, 10, 11, 14 (MISO), 15 (SCK), 16 (MOSI)

  #########################
  #      RESOURCES        #
  #########################

  Written by Kevin M. Smith in 2013.
  Contact: SDI12@ethosengineering.org

  The SDI-12 specification is available at: http://www.sdi-12.org/
  The library is available at: https://github.com/EnviroDIY/Arduino-SDI-12
*/

#define IO4 A5
#define IO3 A6
#define IO2 0
#define IO1 1
#define IO5 2
#define IO6 3
#define IO7 4
#define IO8 5

#define LED 6

bool LEDstatus = false;

const uint8_t sensorPins[] = { IO1, IO2, IO3, IO4, IO5, IO6 };

#include "SDI12.h"

#define SERIAL_BAUD 115200  // The baud rate for the output serial port

// gets identification information from a sensor, and prints it to the serial port
// expects a character between '0'-'9', 'a'-'z', or 'A'-'Z'.
void printInfo(SDI12 * sdi, char i) {
  //digitalWrite(BAT_FAULT, LEDstatus); LEDstatus = !LEDstatus;
  String command = "";
  command += (char) i;
  command += "I!";
  delay(30);

  Serial.print("  --");
  Serial.print(i);
  Serial.print("--  ");

  while (sdi->available()) {
    Serial.write(sdi->read());
    delay(5);
  }
}

// this checks for activity at a particular address
// expects a char, '0'-'9', 'a'-'z', or 'A'-'Z'
boolean checkActive(SDI12 * sdi, char i) {
  String myCommand = "";
  myCommand = "";
  myCommand += (char) i;                 // sends basic 'acknowledge' command [address][!]
  myCommand += "!";

  for (int j = 0; j < 3; j++) {          // goes through three rapid contact attempts
    sdi->sendCommand(myCommand);
    sdi->clearBuffer();
    delay(30);
    if (sdi->available()) { // If we here anything, assume we have an active sensor
      return true;
    }
  }
  sdi->clearBuffer();
  return false;
}

void scanAddressSpace(SDI12 * sdi) {
  // scan address space 0-9
  for (char i = '0'; i <= '9'; i++) if (checkActive(sdi, i)) {
      printInfo(sdi, i);
    }
  // scan address space a-z
  for (char i = 'a'; i <= 'z'; i++) if (checkActive(sdi, i)) {
      printInfo(sdi, i);
    }
  // scan address space A-Z
  for (char i = 'A'; i <= 'Z'; i++) if (checkActive(sdi, i)) {
      printInfo(sdi, i);
    };
}

SDI12 mySDI12;

void setup() {

  pinMode(LED, OUTPUT);
  Serial.begin(SERIAL_BAUD);
  delay(5000);

}

void loop() {
  Serial.println("//\n// Start Search for SDI-12 Devices \n// -----------------------");

  for (uint8_t pin = 0; pin < sizeof(sensorPins); pin++)
  {
    digitalWrite(BAT_FAULT, LEDstatus); LEDstatus = !LEDstatus;
    delay(100);
    pinMode(sensorPins[pin], INPUT);
    mySDI12.begin(sensorPins[pin]);
    Serial.print("Checking pin ");
    Serial.print(sensorPins[pin]);
    Serial.println("...");
    scanAddressSpace(&mySDI12);
    mySDI12.end();
  }

  Serial.println("\n//\n// End Search for SDI-12 Devices \n// ---------------------");

  delay(5000);

}
```

bleckers, 2019-07-30:
Digging a bit deeper, reverting the clock generator code in SDI12_boards.cpp to the previous version (TC4; https://github.com/EnviroDIY/Arduino-SDI-12/commit/764c3b668ef34292363f068bd328f81c4f629f45) creates the same issue with the 48MHz output. But now hot removing the sensor SDI line reverts things back to a working (scanning) state. Plugging the SDI line back in exhibits the same 48MHz output. 

And the SAMD isn't locking up (and USB serial output is working). This doesn't happen with the latest code, the device just completely stops even if you unplug the sensor.

I don't think the 48Mhz signal is coming from the probe either, as it goes away once unplugged (on the probe end).

closed 2019-07-30 by bleckers

SRGDamia1, 2019-07-30:
So, is this something to fix or not?

bleckers, 2019-07-30:
I reopened this issue with something that was slightly related. But it turned out after getting the original issue with the pointers sorted, plugging the Arduino into a daughterboard with level converters caused this related issue (it was outputting a 48MHz signal which I thought might have been related to the clock gen/TC).

As for what caused this original issue. Is it reproducible on other SAMD boards with the examples and the latest SAMD libraries/compilers?

SRGDamia1, 2019-07-30:
I haven't had any problems recently with my Feather M0.

Level converters are often a hassle.