EnviroDIY/Arduino-SDI-12 #55: Support for Arduino Uno WiFi Rev2 (ATMEGA4809)

milenovic, 2019-11-14:
Hello,

Would it be too complicated to add support for the new Arduino Uno WiFi Rev2 which has ATMEGA4809?
There already is a SoftwareSerial library for it here:
https://github.com/arduino/ArduinoCore-megaavr/tree/master/libraries/SoftwareSerial

Unfortunately, adding such support seems to be well beyond my skills.
Could anyone help?

Best,
Milan

SRGDamia1, 2019-11-14:
Did you try using the library as-is?  What was the result?

SRGDamia1, 2019-11-14:
Looking now at the datasheet for the ATMEGA4809, it has a different timer system, so the library will not work as-is.  The timers would need to be re-written.  

If I ever get around to figuring out the ESP8266/32, I'll do this too, but I'm not likely to get to it any time soon.

Note for myself:
  
From datasheet
5x 16-bit Timer (TCA / TCB)

Notes on oscillator options/speeds: (16MHz is 'default' for Arduino)
https://github.com/MCUdude/MegaCoreX#supported-clock-frequencies

millis uses TIMERB0 by default, though selectable B0-B3.
https://github.com/arduino/ArduinoCore-megaavr/blob/master/cores/arduino/wiring.c
In wiring.c:
```c
// the prescaler is set so that timer ticks every 64 clock cycles, and the
// the overflow handler is called every 256 ticks.
```

SRGDamia1, 2019-11-14:
Try using the "delayBase" branch, which uses the ```delayMicroseconds()``` function within the interrupts instead of using timers:  https://github.com/EnviroDIY/Arduino-SDI-12/tree/delayBase.  It will probably work.

milenovic, 2019-11-14:
Hi Sara and thanks for the speedy reply!

Indeed, timers are the main issue. I manged to get v1.1.0 to work now. The only issue there was that __AVR__ is somehow defined for this board, so simply removing all the code that falls in #if defined __AVR__ solved the issue and I got the reading form my SDI-12 sensor!

Now with the version that uses timers, you are right, they are different.
I will try now the "delayBase" branch.
Could you clarify for me what "problems" I can expect from using the old v1.1.0 without the timer part? And if delayBase works, would you recommend it over v1.1.0?

Thanks!

SRGDamia1, 2019-11-14:
The "delayBase" branch uses delayMicroseconds() when reading in or writing out new characters.  It also turns off _all_ interrupts for the entire time that it's reading or writing.  SDI-12 is a really slow protocol (ie, 1200 baud) - sending or receiving a single character takes 3.8ms (1 start + 7 data + 1 parity + 1stop = 10 bits/character, 1 second = 1200 bits).  From the prospective of most processors, that's a really long time to turn off all other computing - absolutely nothing else can happen while the characters are going in or out.

Practically speaking, if you're only doing simple linear operations, ie, talking to one sensor, asking it to measure, waiting for a result, and then writing that to a file, you probably won't have any problems.  You're more likely to start to see problems if you have anything else talking to the board (ie, a UART communication), you're trying to count interrupts (ie, from a tipping bucket), or your SDI-12 sensor is one that returns a large number of values (ie, using "high volume" commands).

SRGDamia1, 2019-11-14:
The "delayBase" version is also trickier to get to work with any other libraries that require interrupt control because the timing is less precise.  So something equivalent to example j isn't likely to work.

dev-itgrapes, 2022-03-20:
Hello,
Is there any update on this issue? Is there anyone that has been able to run successfully this library on the ATMEGA4809 board or have the timer and pin configuration for it?
Thank you very much.