EnviroDIY/Arduino-SDI-12 #58: Example with SDI-12 Sonde

AugustoAlmeida13, 2020-01-28:
I am trying to use your library to develop a datalogger that controls the 6600 V2 probe from YSI. I tried to use the example a from your library, wild card, but got no answers.
I have the following question: My probe is powered by 12 volts, and the power pin provides only 5 volts. Can I feed it externally, excluding the following section of the code that feeds the probe?

` if(POWER_PIN > 0){
    Serial.println("Powering up sensors...");
    pinMode(POWER_PIN, OUTPUT);
    digitalWrite(POWER_PIN, HIGH);
    delay(200);`

PS: If you can do that, is it necessary for the arduino to share the same external power?

Attached is a printout of the Arduino's serial response to the code.

I apologize if the doubts are simple, I have been trying to communicate my probe with the Arduino for a considerable time and I can't.

Sincerely,
Augusto Almeida
[image]

Kevin-M-Smith, 2020-01-28:
Hi @AugustoAlmeida13, 

Does your setup look like this? (Imagine SDI-12 Probe is YSI instead)

https://github.com/Kevin-M-Smith/SDI-12-Circuit-Diagrams/blob/master/basic_setup_usb.png?raw=true

AugustoAlmeida13, 2020-01-28:
> Hi @AugustoAlmeida13,
> 
> Does your setup look like this? (Imagine SDI-12 Probe is YSI instead)
> 
> [image]

I believe so, but shouldn't the ground from the external source be the same as the Arduino's?

Kevin-M-Smith, 2020-01-28:
I would try grounding the sensor to the board as shown (if possible). I don't know the exact details of your setup - but a common ground connection that is closer to the data signal can improve signal quality (compared to relying on a common ground further away).

AugustoAlmeida13, 2020-01-29:
I fed the arduino externally with 12 V and fed the probe to the Vin and GND pins, as you did in one of the links in the examples. It still didn't work, but I believe it is something related to the probe configuration, I will contact the manufacturer and get answers.
Thank you very much for your willingness.

Sincerely,
Augusto Almeida

closed 2020-01-29 by AugustoAlmeida13

Kevin-M-Smith, 2020-01-29:
Hi @AugustoAlmeida13, 

I wasn't suggesting the probe should be powered by 5 volts from the Arduino, I was just suggesting to try connecting the probe's ground to the Arduino GND pin. It is possible the probe requires the full 12 volts to function. 

Best of luck!