EnviroDIY/Arduino-SDI-12 #59: not working with mkr1300

krishnakulkarni88, 2020-04-03:
Hi, I am trying sdi12 with mkr1300 board, The probe is getting detected and when I read the sensor data only battery voltage is coming. but temperature salinity and moisture values are not coming. If the clock was the issue then why the battery voltage is coming. Can you help me how to make It work with mkr1300

krishnakulkarni88, 2020-04-03:
Hi, Thanks for the reply I am attaching the link to my sensor manual(https://www.fondriest.com/pdf/sentek_drill_drop_probe_manual.pdf), I am using 3 sensor-1 probe type. when I try with M commands it was not responding for the sensor values but when I try with C commands it was coming but in between the sensor values I was getting value as 1.

mal62, 2020-04-03:
Hi. Firstly I’ve not successfully got the sdi12 library working with my arduino so there are some bits of the puzzle I don’t have answers to.
Regarding your sensor..... it’s not an easy sdi12 one to interrogate as it is super smart and has lots of different data it can provide and not in a single response message. IE to get temp, moisture and salinity you will need to send a M for temp and then data and then a new m for moisture and then data etc etc.
To start simple ... I suggest you build your program and instead of sending a M and the D just sent the identification instruction.
Ie send 0I! .... the sensor will respond (no need to send a D command) with a character string which should include things like maker, model, serial number etc.

This will confirm your code is good and comms to the sensor is ok.
If you want.. send me your program, although  im no programming expert.

See if you get the 0I! Working and then send back a update.

I’m assuming your sensor is sdi12 address 0.
Also... have you been able to directly interrogate the probe with the software provided?
Does your probe look like the one in the manual on page 1?
Cheers mal