EnviroDIY/Arduino-SDI-12 #60: Corrupted data on the line or the serial print!

tsujido3r, 2020-04-08:
While trying SDI-12 library with 2 Arduino nano boards, one as an SDI-12 slave and another as a Datalogger, I am getting junk values on the device serial monitor while requesting for data (D0!)

I am using the default examples: 
Datalogger - f_basic_data_request
Sensor - h_SDI-12_slave_implementation

[image]

Any idea on what could be causing these errors.

SRGDamia1, 2020-04-08:
Hm.  Do you have any pull-ups on the data line?  Maybe the internal pull-ups aren't enough to overcome noise.  Do you have long cables between the two boards? Can you shorten them?  Do you have any SDI-12 sensors you can test against?  Does it work when talking to one of those?

tsujido3r, 2020-04-16:
I suppose you meant the pulldown, its around 1M ohm. The cable is not long, it's just a jumper cable 10cm in length. That's why I suspect the memory overruns or excessive/incorrect use of String as the main culprit, rather than the line noise. 

Currently, I don't have SDI-12 sensor to test against, sorry. Would it be possible for you to test these two applications against each other(as a sensor and datalogger) in two different arduino dev boards? If you could also verify this, it will be really helpful.

Thanks in advance.