EnviroDIY/Arduino-SDI-12 #62: SAMD21 issue

acutxfarm, 2020-05-06:
I'm using the library to control many SDI12 devices (one for time). For example, using a Sentek drill and drop probe I succed taking data but if I define the WDT as follow, the same SDI commands fail

void set_Low_Power_Clock()
{
  // initialize the WDT watchdog timer
  // Setup GCLK for the watchdog using:
  // - Generic clock generator 2 as the source for the watchdog clock
  // - Low power 32khz internal oscillator as the source for generic clock
  //   generator 2.
  // - Generic clock generator 2 divisor to 32 so it ticks roughly once a
  //   millisecond.
  //   2^(divisor register value + 1)
  // A 32khz clock with a divisor of 32 will then generate a 1ms clock period.
  GCLK->GENDIV.reg = GCLK_GENDIV_ID(2) | GCLK_GENDIV_DIV(CK_DIV);  // 4096 Hz with CK_DIV = 2
  // Now enable clock generator 2 using the low power 32khz oscillator and the
  // clock divisor set above.
  GCLK->GENCTRL.reg = GCLK_GENCTRL_ID(2) |
                 GCLK_GENCTRL_GENEN |
                 GCLK_GENCTRL_SRC_OSCULP32K |     //32.768kHz Ultra Low-Power Internal Oscillator (OSCULP32K) 
                 GCLK_GENCTRL_DIVSEL;
  while (GCLK->STATUS.bit.SYNCBUSY);  // Syncronize write to GENCTRL reg.
  // Turn on the WDT clock using clock generator 2 as the source.
  GCLK->CLKCTRL.reg = GCLK_CLKCTRL_ID_WDT |
                 GCLK_CLKCTRL_CLKEN |          //enable it
                 GCLK_CLKCTRL_GEN_GCLK2;
}

The clock source and channel seems to be unrelated to the one used in the library for the SAMD board, but still I've got a fail, even changing the WDT channel

acutxfarm, 2020-05-06:
Just moved my wdt on another channel, probably the TC3 SAMD21 timer has an internal bug

closed 2020-05-06 by acutxfarm

SRGDamia1, 2020-05-08:
It's been a while since I read up on this, but I'm checking my notes on the which "generic clock generators" 0-3 are used by other common libraries.  My notes say RTCZero uses generator 2; I don't remember what's on 0, 1, or 3 but my notes say they are used by the Arduino core in some way and I'm sure at least one of them is millis() and other essential functionality.  That's why this library uses clock generator 4.  I'm using clock generator 5 for my SAMD watchdog and that seems to work fine.