EnviroDIY/Arduino-SDI-12 #63: Buffer size of 64 is too small for a complete response.

peterj43, 2020-05-07:
I have been using the example code (h_SDI-12_slave and f_basic-data request) and find if I issue a "aC!" command followed by a "aD0" I do not receive all the character in the response. As the maximum length of the values is 75 long <address> + <CRC> + <CR> + <LF> I would expect the max buffer size to be at least 79 chars long. I have tested increasing the buffer  and I now receive the full response. I am performing e2e test on an Arduino MKRWifi1010 and ESP32.

Kevin-M-Smith, 2020-05-07:
Good point, [according to the latest specification](http://www.sdi-12.org/current_specification/SDI-12_version-1_4-Jan-10-2019.pdf), section 4.4, the response to a "Continuous Measurements and Request CRC" command is: 

a<values><CRC><CR><LF>  

where:
- _a_ is a single character
- _values_ has a maximum value of 75 characters. 
- _CRC_ is 3 characters
- CR is a single character
- LF is a single character

suggesting the buffer should accommodate 81 characters.

Kevin-M-Smith, 2020-05-07:
I believe the current value of 64 bytes for the buffer was merely for consistency with the hardware serial buffer limit on the Arduino Uno - I think it should just work to increase it.

Looking forward to the results of your end-to-end tests.

SRGDamia1, 2020-05-09:
PR merged.  I'm leaving this open until I add the if's that I want around it.