EnviroDIY/Arduino-SDI-12 #65: CRC Implementation

peterj43, 2020-05-08:
From what I can see the CRC has not been implemented. Being very new to SDI-12, is this because no one uses it or is it some thing that need implementing? My current project is developing a SDI-12 version of our sensor so could do to know if I will need to add this functionality?

SRGDamia1, 2020-05-09:
_I've_ never tied to use the CRC functionality.  I don't even know if the instruments I have support it.  I don't know if anyone else uses it; no one's asked about it.  I guess if you're really worried about avoiding communication errors you could try to implement it.

If you do decide you really want to get the CRC working, you shouldn't have any trouble finding a C++ example of a CRC calculation.  I can review a pull request.

neilh10, 2021-07-13:
Hey great to see a CRC covering the responses.

Sharing some research I've done - reference https://sdi-12.org/specification SDI-12 Version 1.4, January 30, 2021 
4.4.12 Requesting a Cyclic Redundancy Check (CRC)
To be version 1.3 (or higher) compliant, the sensor must support CRCs
Specificies the algorithim

From my old copy "CTD Integrators Guide.pdf" Decagon Devices CTD-5 & CTD-10 Sensor - 
http://publications.metergroup.com/Integrator%20Guide/CTD%20Integrator%20Guide.pdf
"After the power up time, our sensors are fully compatible with all the commands listed in the SDI-12
Specification v1.3 except for the continuous measurement commands (aR0 – aR9 and aRC0 – aRC9)"
and they have the algorithm they use.

Insitu SDI-12_Commands_Tech_Note 2020-05-22.pdf
Identifies SDI-12 V 1.3 Command Set