EnviroDIY/Arduino-SDI-12 #73: NodeMCU Compatibility

KareemWaheed, 2020-10-22:
Hi,

first of all thank you for the great work put inside this library.

I'm trying to use the library with NodeMCU module I successfully read from a sensor using arduino uno, but while connecting the nodemcu to the sensor and flash this code : 

```#include <SDI12.h>

#define SERIAL_BAUD 9600 /*!< The baud rate for the output serial port */
#define DATA_PIN 5         /*!< The pin of the SDI-12 data bus */
#define POWER_PIN 4       /*!< The sensor power pin (or -1 if not switching power) */

/** Define the SDI-12 bus */
SDI12 mySDI12(DATA_PIN);

/**
  '?' is a wildcard character which asks any and all sensors to respond
  'I' indicates that the command wants information about the sensor
  '!' finishes the command
*/
String myCommand = "?I!";

void setup() {
  Serial.begin(SERIAL_BAUD);
  while (!Serial)
    ;

  Serial.println("Opening SDI-12 bus...");
  mySDI12.begin();
  delay(500);  // allow things to settle

  // Power the sensors;
  if (POWER_PIN > 0) {
    Serial.println("Powering up sensors...");
    pinMode(POWER_PIN, OUTPUT);
    digitalWrite(POWER_PIN, HIGH);
    delay(200);
  }
}

void loop() {
  mySDI12.sendCommand(myCommand);
  delay(300);                    // wait a while for a response
  while (mySDI12.available()) {  // write the response to the screen
    Serial.write(mySDI12.read());
  }
  delay(3000);  // print again in three seconds
}```

the serial log return : [console log](https://i.imgur.com/cCRecFn.png)

could someone help with this

Thank you

SRGDamia1, 2020-10-27:
Hm.  Exactly the same program works on the Uno, right?  The code looks right.

Are you shifting voltage?  The Uno is 5V but NodeMCU (ESP8266) is only 3.3V.  SDI-12 standard is 12V, but manufacturers use other voltages as well.  Voltage shifters themselves can also cause issues if they don't have a fast or clean enough response.  Are you sure you have the right pins?  

What kind of sensor is it?

KareemWaheed, 2020-11-06:
> Hm. Exactly the same program works on the Uno, right? The code looks right.
>
> Are you shifting voltage? The Uno is 5V but NodeMCU (ESP8266) is only 3.3V. SDI-12 standard is 12V, but manufacturers use [...]

I'm using Teros sensors from metergroup
actually, they are working right now sorry for the delay in response I think it was wiring problem

Thank you for your contribution to making this library

SRGDamia1, 2020-11-09:
Ok. I'm glad it's working!

closed 2020-11-09 by SRGDamia1