EnviroDIY/Arduino-SDI-12 #76: SDI-12 Data error

nsecgo, 2021-05-13:
[image]
[image]

nsecgo, 2021-05-13:
[image]

SRGDamia1, 2021-05-13:
What's the error?

nsecgo, 2021-05-14:
> What's the error?

[image]

nsecgo, 2021-05-14:
May be caused by the quality of the DuPont wire
The error no longer appears after replacing the Dupont wire

closed 2021-05-14 by nsecgo

SRGDamia1, 2021-05-14:
Oh, sorry, I didn't even notice those.  This library is "bit-banging" for the serial port, so occasional blips in the serial string are more common than is ideal.  But low quality wires will definitely make it worse.

nsecgo, 2021-05-19:
@SRGDamia1 I found that the longer the ground wire from Arduino's GND to External power supply's GND(The orange line in the image above), the more frequent the errors are. 
Any ideas are greatly appreciated.

SRGDamia1, 2021-05-19:
I'm not really great with hardware.  I don't have any suggestions other than to use the best quality wire and connections you can and don't have wires any longer than necessary.

neilh10, 2021-05-19:
Ground issues are really tricky. Usually its about having good design practices. 
 Try keeping the ground/power wires as short as possible and keep the two power supplies +5V and +12V  together off the same plug.  Possibly get a +5V regulator and power it off the +12V. The +5V would power the mayfly 
 
Part of the problem may be figuring out just where is the problem.  There is "ground bounce", and then there is noise on the data line. Both could result in corrupted data as seen by the Mayfly.  
IMHO the Mayfly SDI-12 description of connecting data wire directly to the processor is really "proto-typing", doing something to be able to get the software to work under ideal conditions.  The  SDI-12 specification has a defined  wire and voltage interface that has more latitude for noise.  Ground bounce is a much more technical problem.