EnviroDIY/Arduino-SDI-12 #85: SDI12 - ESP32 NOT OPEN BUS

matiasAS, 2022-02-04:
Hello everyone, I am trying to read data from a Campbell Scientific sensor by sdi12 with an ESP32, but the sensor does not recognize me or it simply does not detect that the sdi12 object is available.
I share my code in case someone could help me

I'm from Chile, that's why there are some lines in Spanish. If there was someone who spoke my language much better!!!

//#include <SPI.h>
//#include <UIPEthernet.h>
//#include <SPI.h>
//#include <Ethernet.h>
#include <WiFi.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>

#include <WiFi.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include "SDI12.h"
SDI12 mySDI12(2);
String datos = "";
 bool isAvailable(int pin){
 // SDI12 mySDI12(pin);
 // Serial.println("DATAPIN: "+this->DATAPIN);
  return mySDI12.available();
}
void printBufferToScreen(int pin){
  Serial.println(" en print buffer screen ");
  String buffer = "";
  buffer +="{\"H\":\"";
  byte cont = 0;
  byte largo = 0;
  byte cont_largo = 0;
  mySDI12.read(); // consume address

  while(mySDI12.available()){
    Serial.println("dentro del while print buffer screeen");
    char c = mySDI12.read();
    if(c == '+'){
      if(cont == 0){
      }
      if(cont == 1){
        buffer += "\",\"CE\":\"";
      }
      if(cont == 2){
        buffer += "\",\"T\":\"";     
      }if(cont == 3){
        buffer += "\",\"PDIEF\":\"";
      }
      if(cont == 4){
         buffer +="}";
      }
       cont++;   
      if(c == '-') {
        buffer += '-'; 
        }
      if(c=='\0'){
      }
    } 
    else {
      buffer += c;        
    }

    delay(100);   
    }
  Serial.println("despues del while en print buffer screen");
  datos = buffer;
}

void takeMeasurement(char i,int pin){
  Serial.println(" take measurement");
  String command = ""; 
  command += i;
  command += "M3!"; // SDI-12 measurement command format  [address]['M'][!]
  mySDI12.sendCommand(command); 
  while(!mySDI12.available()>5); // wait for acknowlegement with format [address][ttt (3 char, seconds)][number of measurments available, 0-9]
  delay(100); 
  mySDI12.read(); //consume address
  
  // find out how long we have to wait (in seconds).
  int wait = 0; 
  wait += 100 * mySDI12.read()-'0';
  wait += 10 * mySDI12.read()-'0';
  wait += 1 * mySDI12.read()-'0';
  
  mySDI12.read(); // ignore # measurements, for this simple examlpe
  mySDI12.read(); // ignore carriage return
  mySDI12.read(); // ignore line feed
  
  long timerStart = millis(); 
  while((millis() - timerStart) > (1000 * wait)){
    if(mySDI12.available()) break;                //sensor can interrupt us to let us know it is done early
  }
  
  // in this example we will only take the 'DO' measurement  
  mySDI12.flush(); 
  command = "";
  command += i;
  command += "D0!"; // SDI-12 command to get data [address][D][dataOption][!]
  mySDI12.sendCommand(command);
  while(!mySDI12.available()>1); // wait for acknowlegement  
  delay(300); // let the data transfer
  //Serial.println("antes de printBufferToScreen()"); 

  printBufferToScreen(pin); 
  mySDI12.flush(); 
}

  String obtenerDatos(int pin){
    float temp_c;
    float humidity;
    temp_c = 0;//sht1x.readTemperatureC();
    humidity = 0;//sht1x.readHumidity();
      takeMeasurement('0',int(pin));
// datos +="\",\"PH\":\"\","+"\"TA\":"+"\""+temp_c+"\","+"\"HA\":"+"}";
  
  datos +="\",\"PH\":\"\",";
  datos += "\"TA\":";
  datos +="\"";
  datos += temp_c;
  datos +="\",";
  datos += "\"HA\":";
  datos +="\"";
  datos += humidity;
  datos +="\"";
  datos += "}";
  int largo_datos = datos.length()+1;
   char datos_array[largo_datos];
   //char nuevo_datos_array[largo_datos];
   String datos_limpios = String("");
   datos.toCharArray(datos_array,largo_datos);
   for(int i=0;i<largo_datos;i++){
    if(datos_array[i] != '\0' && datos_array[i] != '\n' && datos_array[i] != '\r'){
      datos_limpios+=datos_array[i];
    }
   }
  return String(datos_limpios);
  }

String jsonSensor(int id){
  Serial.print("id: ");
  Serial.println(id);
  String data = obtenerDatos(id);
  String json = "{";
  json += "\"pin\":";
  json += "\"";
  json += id;
  json += "\",";
  json += "\"available\":";
  json += "\"";
  json += isAvailable(id);
  json += "\",";
  json += "\"data\":";
  //json += "\"";
  json += data;
  //json += "\"";
  json +=  "}";
  return String(json);
}  

   
void setup() {  
    Serial.begin(115200);
    delay(500);
    mySDI12.begin();
}

void loop() {
  Serial.println(jsonSensor(2));
  delay(1000);
}

I hope you can help me

Thank you
Greetings
Matias

matiasAS, 2022-02-04:
I can get data with arduino menga and old version sdi-12 library (incompatible with esp32)

matiasAS, 2022-02-24:
Hello @ALL @majenkotech  @diegogmx @aufdenkampe @Kevin-M-Smith , could you help me please, I've been waiting 20 days for an answer

matiasAS, 2022-02-24:
@SRGDamia1

diegogmx, 2022-02-25:
> Hello everyone @majenkotech <https://github.com/majenkotech> @diegogmx

Hola Matías, perdón, no vi el mensaje, tampoco estoy particularmente
involucrado en el desarrollo del proyecto, los problemas que he tenido con
estos sensores generalmente se deben a temas de timing, particularmente con
el pulso inicial de sdi12 y con el tiempo de espera de respuesta, si tenes
un osciloscópio a mano podrías hacer unas capturas y vemos que sucede ahí

matiasAS, 2022-02-25:
Hola @diegogmx gracias por tu respuesta, y que bueno poder hablar en español, jajaja, soy de chile. He estado preguntando en grupos de electrónica en facebook y me dicen que debo pasar la señal de sensor por un conversor para bajar de 5v a 3.3v que es a lo que trabaja el ESP32. 

La razón de usar un ESP32 primero por su precio y segundo porque se puede conectar a wifi y es super estable y confiable en ese sentido. Ya tengo un plan B que sería un arduino mega con el shied YUN, con eso estaría super!!. 

Sobre lo del osciloscopio....tengo uno que compré en aliexpress, es uno mini, pero la verdad que en los 7 años que estuve en la U, una de las cosas que más de una vez vi y que nunca aprendí a usar es....un osciloscopio.....

Saludos

matiasAS, 2022-02-25:
ahora @diegogmx mi duda es si en el readme de este repo aseguran compatibilidad con ESP32, en caso de tener que usar algun tipo de conversor lo deberia decir o estoy equivocado?

diegogmx, 2022-02-25:
Yo lo use con un samd21, realmente no se lo de la compatibilidad del esp,
lo de la señal de 3v3 podría llegar a ser pero me inclino a decir que no,
yo lo use con 3v3 directo y sin problema, depende de los niveles efectivos,
lo que si es que perdés inmunidad a ruido, lo que te sugeriría es, si vas a
seguir en electrónica aprende ayer a usar un osciloscópio, te puede ahorrar
horas y horas de debug, después, si mal no me acuerdo, la biblioteca tiene
un modo que te permite hacer un dump en el puerto serie, así ves que es lo
que está pasando por el bus, a falta de osciloscópio proba eso, otro
detalle, tenes que usar un pin con funcionalidad de pwm para este coso,
verifica eso también

matiasAS, 2022-02-25:
@diegogmx sinceramente nosé de lo que me hablas....y no, no soy electrónico. Lo que tengo implementado es obtener datos con arduino, transmitir por mqtt a una raspberry y ahi con python/django hacer todo lo de transmitir los datos a un server central, almacenar datos y mostrar en gráficos la data