EnviroDIY/Arduino-SDI-12 #86: Watchdog Problem SAMD21

nunomiguelferreira, 2022-02-25:
Hello,

I'm interfacing with a TEROS 12 Soil Moisture Sensor, and whenever I activate the watchdog in my SAMD21 processor (based on Feather M0), the system hangs on reading data (with 8s timer). 

The same code without the watchdog activation works perfectly, so I guess it must be a conflict between the library timers and the watchdog ones?

I would like to know any suggestions for this issue.

Thank you for your time.

neilh10, 2022-02-25:
@nunomiguelferreira fascinating that you have the ArduinoSDI12 working on the SAMD21/Feather M0.   Is it with ModularSensors? or another framework. 
I guess my interest is how the internal Arduino-SDI-12  1200 sampling is run within a SAMD21. This could impact any watchdog enabling, so just curious.

nunomiguelferreira, 2022-02-27:
At the moment I'm just doing a simple bus read (based on continuous measurement example) to get the 3 values the TEROS-12 provides. Haven't moved on to the Modular Sensors Framework.

The values seem correct from internal testing, but whenever the watchdog is enabled, I suppose there's maybe a conflict with the timer used for the library and watchdog.

As for the PCB, we use the FeatherM0 pindefs, but its actually a custom PCB, but I guess that doesn't really matter in this case. The major issue with deploying this to the field will be the case of hangup... there is the need for a watchdog for system recovery, so if I can't make it work with watchdog, it might be a bit worthless.

neilh10, 2022-02-28:
@nunomiguelferreira Good to hear the basics are working . Thats what I tend to do - verify an instruments interface in standalone mode as the first step, and then add functionality incrementally.
 Sounds like its an Arduino framework and the code is using TC3 which works from initialization. 
You may want to check you following SDI12_boards.cpp #207 (and from #249)and ensure you aren't trampling on any of the interrupts for TC3

`// Arduino Zero other SAMD21 boards
//
#elif defined(ARDUINO_SAMD_ZERO) || defined(ARDUINO_ARCH_SAMD) || \
  defined(__SAMD21G18A__) || defined(__SAMD21J18A__) || defined(__SAMD21E18A__)

void SDI12Timer::configSDI12TimerPrescale(void) {`

closed 2022-03-17 by nunomiguelferreira