EnviroDIY/Arduino-SDI-12 #94: Documentation missing?

WestfW, 2022-07-09:
It says: "Full details on the library functionality can be found on github pages: https://envirodiy.github.io/Arduino-SDI-12/ "
But that loops back to the same https://envirodiy.github.io/Arduino-SDI-12/ page, and it seems to only contain info about history and supported chips and conflicts and limitations, with no information on how to actually USE the library...

I don't see any usage documentation in the source repository either, although there might be some Doxygen scripts that are supposed to extract it.

SRGDamia1, 2022-07-12:
Did you look at the examples?

SRGDamia1, 2022-07-12:
The https://envirodiy.github.io/Arduino-SDI-12/class_s_d_i12.html page has details about every function of the SDI-12 class, which is the only one meant to be used externally.  If you use the "SDI-12" dropdown in the upper right of the documentation, you can jump to any section of that page.

There's also a drop-down in the upper right to help jump to whatever example you might want.

WestfW, 2022-07-17:
>  If you use the "SDI-12" dropdown

Ah!  There it is!  yes, that has all that I was looking for.
I think it's a bit odd that there isn't a link in the text of the page (given the many other links), but I'll concede that it's all there once I notice that there ARE drop-downs at the top of page.