EnviroDIY/Arduino-SDI-12 #141: i am using hydros21 sensor which is working on sdi12 protocol and currently i am stuck in a issue which is sensor is not working properly and not given data correctly so any one help me for this

praveerkr, 2024-06-03:
i am using hydros21 sensor which is working on sdi12 protocol and currently i am stuck in a issue which is sensor is not working properly and not given data correctly so any one help me for this

SRGDamia1, 2024-06-03:
I'm sorry, I can't help you without more details about what the sensor is and what is not working properly.

praveerkr, 2024-06-05:
i am new to arduino, i am working on a project to use arduino to collect data from several SDI 12 sensors, for now i am trying to figure out how to view the results from the sensors on on my PC. tried using the SDI 12 library from https://github.com/EnviroDIY/Arduino-SDI-12 but can’t get it to show the results from the sensors. Please help .I will try running the address change example, as for now i am trying to use a simple arduino board (arduino uno). the main idea of my project is to use an arduino board to collect the data from an SDI-12 based sensor and then transmit it to another for storage. Do you think it will be possible to connect about 1 sensors to the arduino board and collect the data or an extra hardware is required to do that?

Thanks