EnviroDIY/Arduino-SDI-12 #149: ESP32C6 Compatibility

ndoo, 2025-01-06:
ESP32C6 requires NEED_LOOKAHEAD_ENUM but the existing ifdefs in SDI12.h skip it.

It seems like the ESP-IDF version ESP32C6 Arduino framework uses for my specific ESP32C6 board (Seeed XIAO ESP32C6) is above the version in SDI12.h resulting in NEED_LOOKAHEAD_ENUM not being set and compilation failing due to the enum being missing (similar to ESP8266/ESP32 build failures before adding support).

Is there a rationale for the existing ESP IDF version check? Is there a more portable/universal check?

My build environment in PlatformIO:
```
[env:xiao-esp32c6]
platform = https://github.com/mnowak32/platform-espressif32.git#boards/seeed_xiao_esp32c6
platform_packages = 
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.2
    framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.2/esp32-arduino-libs-3.0.2.zip
framework = arduino
board = seeed_xiao_esp32c6
```

SRGDamia1, 2025-01-24:
That enum is set in the current core: https://github.com/espressif/arduino-esp32/blob/master/cores%2Fesp32%2FStream.h#L40

It was added for release 3.0.5/IDF 5.1.4. When I set up the define, I don't think I realized that versions 3.0.0 - 3.0.4 use the same IDF version without having the enum. It looks like there are also defines for the core  that I could use as the check here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp_arduino_version.h. I don't know when I'll get to applying that change.

SRGDamia1, 2025-02-18:
This should be fixed by 5e1b0545c8f2421f46c4f62620323eca981d7e4d

closed 2025-02-18 by SRGDamia1